bigtreetech/BIGTREETECH-S42B-V1.0 #1: Manual file

schpuntik, 2020-01-07:
Manual file is brocken

yangwenxiong, 2020-01-08:
> Manual file is brocken. Please upload again

Have been updated

yangwenxiong, 2020-01-08:
[https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/1#issue-546557873](url)

yangwenxiong, 2020-01-08:
[image]

yangwenxiong, 2020-01-08:
[image]

yangwenxiong, 2020-01-08:
[image]

yangwenxiong, 2020-01-08:
[image]

closed 2020-03-23 by bigtreetech