bigtreetech/BIGTREETECH-S42B-V1.0 #3: Motor Spin after M17 command

ehtnevets, 2020-03-15:
After auto-disable timeout, sending M17 (enable) will cause motor to jump to almost the end of travel of an axis. Is there work around?

Thanks

sae74, 2020-04-30:
I have exactly the same problem in rotation after shutdown

ehtnevets, 2020-05-01:
I'm ordering the MKS closed loop board. I'll let you know if MKS brand exhibits the same problem, in 2-3 weeks. Shipment has been really slow from China to U.S.

sae74, 2020-05-01:
I will wait for feedback from you.
Thank!

ehtnevets, 2020-05-21:
I ordered MKS from AliExpress. It hasn't arrived after more 2 months. I disputed this with AE but seller (MKS) keeps extending protection program. AE only allows dispute if the protection program is expiring in a day. The computer generated response from AE keeps saying my dispute is invalid. So, I call my bank to charge back.  I gave up MKS.

Anyway, I gave up with this closed loop stuff as it doesn't have the advanced smoothing feature like Trinamic drivers. It's loud and vibrates my print bed. Lowering the current seems to help but sacrifices max speed. Even after I found all the disable motor-off feature in Marlin and Klipper firmwares to avoid the jump from re-enabling, I think this closed loop solution gives worse bottom line result.

To top it off, the magnet always gets loose. I used all different type of glues, cement, superglue, epoxy, etc. They don't help. Especially, the ones I used in the extruders with linear/pressure advance feature (linear advance causes the motor to spin back and forth very fast and for short distances, aka violent jerking).  

So, in the end I realized that closed loop motor this cheap ain't cut it. If you want to go closed loop, get one with $100s/axis. Plus, if your non-closed loop stepper is skipping steps, there is other fundamental issue that needs to be addressed. But for sure not going with closed loop.

Back to Trinamic.

sae74, 2020-05-21:
Thank. But I need a device not for a 3D printer.
It’s very important for me to keep track of the number of steps.

ehtnevets, 2020-05-21:
Perhaps, you should use the more expensive version (about 3x more). They use more expensive components even down to the passive ones like bigger capacitor, tighter tolerance. 

When it comes to closed loop control, I re-learned (I started a company that designed and manufactured robots for semiconductor industry) that you get what you pay for.  

You can always keep track your steps with the software. The software is the one doing the stepping anyway.

ehtnevets, 2020-05-21:
And if your usage include moving the motor with a lot of high accel and decel, the magnet could be another problem.

sae74, 2020-05-22:
I need to control the steps for a water pump system. Need high accuracy +/- 0.005 ml liquid

ehtnevets, 2020-06-15:
I finally got my MKS boards but decided that the motion smoothness is more important when it comes 3D printing and decided not to bother with it. 

I have 1 board set with OLED and 4 just board set that I want to sell. If you are in US, I can sell them to you at cost + shipping. Cheaper than you find on eBay.

sae74, 2020-06-16:
Thank.
But I am in a different country.
If you know what Chernobyl is, or you watched the Chernobyl series, then this is about my country, Ukraine.

ehtnevets, 2020-06-16:
Ukraine. Remind me of my a friend who told a story about Chernobyl cleanup when he was in the Army there.

Quas7, 2020-09-10:
I experience the same issue for my FDM extruder after powerdown and enabling again. It always stripped my filament as it retracted with max speed x100m of filament. ;)

Simple solution was to disconnect the enable-pin from the cable-connector and bridge the enable signal on the stepper PCB to GND (active low) with a dupont wire.
This way, the stepper driver stays of course always active but there is also no "reset to 0" at the start of the print.

Quas7, 2020-09-10:
I flashed today the latest firmware out of this repo to adjust the PID values a bit and now the enabling/disabling of the stepper driver does not result in any erratic spinning anymore.

I guess, the issue got solved by the devs before uploading the code here.

sae74, 2020-09-11:
How did you upload the firmware?
Where did you connect the programmer?

Quas7, 2020-09-11:
Took the pinout from here: https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/blob/master/Item-Pinmap.PDF
[image]
Read carefully the pins as there are two rows of pins (OLED and aux. connector) and one column of pin descriptions depicted

SWCK, SDO, GND are required - you can also disconnect the board from the supply and plug in the 3v3 from the programmer.
Make sure, the board is not "enabled" for flashing as this seems to result in issues (seems to be EMI as my OLED also shows artefacts from time to time).
I used a st-link v2 clone but any STM32 programmer will do, I think.

Quas7, 2020-09-22:
https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/16
I will continue to post results for optimizing the communication frequency.

AbeFM, 2020-09-30:
> Read carefully the pins as there are two rows of pins (OLED and aux. connector) and one column of pin descriptions depicted
> 
> SWCK, SDO, GND are required - you can also disconnect the board from the supply and plug in the 3v3 from the programmer.
> Make sure, the board is not "enabled" during flashing as this seems to result in issues (seems to be EMI as my OLED also shows artefacts from time to time).
> I used a st-link v2 clone but any STM32 programmer will do, I think.

I hooked up those 3, plus 3.3V, pin for pin over to my generic STLink2 dongle. The dongle mounts in Windows, but beyond that, I can't get anything to talk. Should it show up as a com port? Do I need a driver for the motor controller board (S42B) itself?

Do you upload firmware with the SerialPort Assistant, Arduino, VS, or what?

I have a feeling I have a bad board, but trying to see if I can get it talking. It holds position, calibrates, works with the screen - but won't register steps from printer. I even recompiled to get the longer pulse widths, still a no go.

Thanks - sorry for the off topic post - perhaps having the process more explicitly laid out will help others in the future.

Quas7, 2020-09-30:
depending on the dongle you have, it should show up as a generic USB device "STM32 STlink" or as a "ST-Link Debug" (STLink v3) with an additional COM-port. 
If you have a Baite dongle it shows up as a com port via "STM32 STLink".

Do not connect the 3V3 parallel to powering the board itself. There can only be one supply voltage and you could damage either supply (most likely the dongle, if it is a cheap clone).

For coding, I work with platform.io in as a plugin from the market inside visual studio code (https://code.visualstudio.com/).
See here for installation: https://platformio.org/install/ide?install=vscode

You can upload the binary via visual studio itself if you configured your dongle correctly in the platformio.ini via upload_protocol.
Youtube explains it in more detail here for instance: https://www.youtube.com/watch?v=dany7ae_0ks&ab_channel=MakeUseOf

good luck.

AbeFM, 2020-09-30:
Thanks for the detailed response!

1) I have the printer shut off, and the 3.3 line from the programmer
(programmer powered from usb) powering everything. Is this ok?

2) I've had the OLED screen plugged in - is that an issue?  I put the board
in Open Loop mode (sw3 OFF) during this.

3) In Device Manager, under "Universal Serial Bus devices" I have "STM32
STLink", but I have no entries under "Ports" as I would expect with a
serial adapter plugged in. Do you feel that is because I don't have a
driver for the S42B on the computer, or because I may not have fully
installed the STLink?

Watching videos and getting PlatformIO installed.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2020-09-30:
Ah, ok. So I didn't have 5V run to the driver's socket, so the little adapter board wasn't sending said +5v to the S42B. I checked that I could talk to another SM32 board (one of those Arduino Nano looking deals). Your pinout was helpful.

Now with that jumpered, I was able to flash your firmware onto it. Everything works including printer commanded moves. I just need to remove the jumper and Say It With Solder and I may actually be able to print stuff again. Thanks for all the help!!!!

Does that have the bug fixed with the jumping when "enabled" is un-asserted?

AbeFM, 2020-10-02:
Still sufferings inconsistent connections. Sometimes it doesn't work without being plugged into printer, sometimes it doesn't work IF it's plugged into the printer, I'm getting a replacement programmer but I suspect it's the board. Aside from that, I need to change my PID values, and had some questions for @Quas7 :

> Make sure, the board is not "enabled" during flashing as this seems to result in issues (seems to be EMI as my OLED also shows artefacts from time to time).

Can you explain what you mean here - you mean the enable pin from the printer being asserted? So the bad state is dependant on the firmware settings? I've had the board unplugged from the motor and oled when programming.

I don't think I have the same issue as you, mine gets the first line (initializing oled) then resetting, so it's just flashing that message at ~2 Hz.

> changed in main.c the D value for the PID loop to improve a tiny bit the extrusion performance:
> int16_t kp=30;
> int16_t ki=10;
> int16_t kd=200; //was 250

Have you used this for any movement axis? I was getting run away oscillations on my CR-10 and wondering what good starting numbers might be.

> Warning! contains reset-to-zero-bug after re-enabling the driver: [s42b_org_as_delivered.zip]

Can you explain what that error is so I can verify it's removed from my firmware?

Quas7, 2020-10-02:
@AbeFM 
I had also many connection problems in the beginning but quickly noticed, that this happens, if the board was powering the motor (driving the PWM stage). 
Disconnecting the motor did not help in my trials likely as the PWM chip will still try with maximum motor voltage to get current pushed into a non existing motor. These voltage peaks seem to still interfere with the programming communication.
Just to be clear, I would not go that far to say that this is a design flaw of the PCB as on such a small board this might be really hard to prevent. BTW, the PWM frequency seems to be 187kHz according to Jans calculations.

You can set the "enabled active high or low" via the menu to switch the driver stage off during programming. 

My PID settings are only for my extruders and naturally they feature a quite heavy dampening that prevent oszillations. 
If you want to do a heuristic calibration, I would recommend the Ziegler-Nichols methode for PID systems as a starting point (https://en.wikipedia.org/wiki/Ziegler%E2%80%93Nichols_method).
You will need to generate a stable oszillation with a high P and I=D=0 and then measure the oszillation frequeny with a smarthone app like "vibration analysis" (android). 
The rest is then following the recipe on wikipedia and I would start with the derived values for "classic PID".

AbeFM, 2020-10-05:
Ah, thanks again for a good answer. I'm unfortunately reasonably versed in PIDs, but have never had a case I COULD just tune like that. I guess the problem is you're reading the pulley, not the stage, probably making things worse. Anyway, I'll try it when I managed to get it flashed - even if I didn't need to I've always wanted to use the phone accelerometer for something useful.

Ah, ok, thanks for the answer: Making sure it's not enabled. Which pretty much means "does motor freewheel", if it's holding it, you'll know.

It would be awesome if (along the lines of "it's not a design issue) they let you flip a DIP switch to get into reliable programming mode. :-)

swanepoeljan, 2020-10-07:
> It would be awesome if (along the lines of "it's not a design issue) they let you flip a DIP switch to get into reliable programming mode. :-)

If they just broke-out the BOOT0 pin of the MCU to the DIP switch (or some extra jumper) we could have used the internal system bootloader to easily flash new firmware with only a normal USB to Serial bridge...