bigtreetech/BIGTREETECH-S42B-V1.0 #4: Configuration for 0.9 degree step angle steppers

WayeeCool, 2020-04-25:
I have steppers with a 0.9 degree step angle instead of the 1.8 degree step angle the S42B seems configured for.  On the MisfitTech Nanosteppers, that these are based on, support for 0.9 degree steppers was added a long time ago and I am wondering if the manual for the S42B just leaves out how to configure for 0.9 degree steppers.

Andrei43D, 2020-07-25:
I have 0.9 steppers too. Interested, if its possible to use this board with existing TMC2209 drivers and 0.9 steppers. BigTreeTech, I'm up for testing, of you are interested. This could become a really useful addition to almost any printer, if the board could work with different motors and different drivers (like 2209 or 2226)
Keep up your great work!)

setar, 2020-11-07:
very easy fix
https://github.com/setar/BIGTREETECH-S42B-V1.0/commit/cead5cab6ba368c2d2e6fd1c4872c61c3020dbd4

caesar1111, 2020-11-12:
@setar : which 0.9 degree stepper you are running for your for? And can you still confirm they run without flaws?

setar, 2020-11-12:
it work for me now http://roboforum.ru/forum107/topic18565.html