bigtreetech/BIGTREETECH-S42B-V1.0 #6: [FR] TMC stepper driver

Protean-Man, 2020-06-16:
What you have done is impressive with the integrated closed loop circut - but the A4950 is older and antiquated. What about the TMC4671 - ? Or other even two options with the TMC5160 & TMC2209?

frank-pian, 2020-06-29:
These closed-loop drives are based on 'Nano Zero Stepper'.
This code does not use the TMC chip.

dzid26, 2020-07-02:
> What you have done is impressive with the integrated closed loop circut - but the A4950 is older and antiquated. What about the TMC4671 - ? Or other even two options with the TMC5160 & TMC2209?

This would be close loop control on chip instead of software. 

They actually sell (expensive) close loop steppers. https://www.trinamic.com/products/pandrive-smart-motors/#c2483

garkin, 2020-07-28:
This whole assembly is less than $20 atm. What makes A4950 antiquated in this price range? 
What exactly your change adds to the plate? How much does it costs? 
What ammount of work required to update firmware?
[Mouser shows $30 per item](https://ru.mouser.com/Search/Refine?Keyword=TMC4671) 
[TMC4671-LA datasheet](https://ru.mouser.com/datasheet/2/256/TMC4671-LA_datasheet_Rev2.00-1880648.pdf)

Protean-Man, 2020-07-28:
In the usage of stepper drivers the A4950 compared to others, it is older and antiquated. This closed loop system is great and yes it is at a great price range for what you get. 

My question still stands given that they could make create this for $20, why wouldn't they want to sell a version with more advanced drivers? 
The closed loop systems from TMC are very expensive - I'd bet Bigtreetech could do it for much cheaper and get similar quality.

fake-name, 2020-08-05:
The TMC4671  is completely irrelevant here, because the servo controller is implemented in the STM32 MCU. Sure, you could replace the entire circuit with another, but at that point, it's just a different product.

> In the usage of stepper drivers the A4950 compared to others, it is older and antiquated.

The A4950 is not a stepper driver, and it is not antiquated. It's a perfectly servicable modern single-chip current limited H-Bridge IC. 

If you think the TMC4671 is a replacement or upgrade for the A4950 (and the way it's being used), you have no idea what you're talking about. 

The entire point of doing the servo loop in a inexpensive MCU with a simple H-Bridge output stage is that it's inexpensive, and you can compensate for any hardware nonlinearities with clever software. The TMC4671  is the opposite of that.

dash3811, 2020-11-18:
If you want to use upgraded drivers for closed loop then you should just buy a SKR 1.4 or 1.4 Turbo and instead of connecting the wire to the adapter driver just connect it to the spot in front of the drivers labeled x-cls, y-cls, etc. That's the purpose of those pins.