bigtreetech/BIGTREETECH-S42B-V1.0 #7: OLED Screen doesn't stay on

CircuitSetup, 2020-07-27:
Is there a reason the OLED screen would flicker and turn off completely when the stepper motor is enabled? The screen never seems to stay on. It does stay on when the motor is not enabled, but not always.

cofhal, 2020-07-31:
I have the same problem.

CircuitSetup, 2020-08-04:
@bigtreetech - any ideas on this?

charchess, 2020-08-13:
Same here.
to be more specific, I got 4 motors and so far :
* the X one is working fine
* the Y one is dancing (when a command is issued he is going back and forth around the target position I guess, not related to that ticket, just for information)
* the Z1 and Z2 are just not responding and display is shuting down after a couple of seconds or sometimes flickering before shuting down

I tried to invert X and Z1 motor and Z1 motor still doesn't repond properly and still have that shutdown issue.

when moving a bit the oled screen, I got the impression there is some bad contact somewhere, will investigate further in the next days and see if I can find anything.

For those having the same problem (screen shuting down), can you control it (just some display problem) or is the motor not obeying the commands ?

CircuitSetup, 2020-08-13:
@charchess make sure the current and steps are set correctly, and after doing so, run the calibration.

When my screen goes out the motor still responds without issue. If I want to change something I have to do it before the motor is activated though. I swapped with a similar screen, and another driver and it didn't make a difference.

kookaburra88, 2020-08-14:
Same issue on my side with 2 motors + 2 OLED screens

kamilj, 2020-09-01:
More and more often I think about creating a database for Chinese products that are just a scam. Products of such poor quality are practically absent in Europe or the USA. I have the same problem in 3 drivers.

charchess, 2020-09-02:
@CircuitSetup my issue on the motor was ..I did misread the documentation ...
https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/blob/master/42%20Stepper%20Motor%20Closed%20Loop%20Driver%20Control%20Board%20Manual.pdf page 20 explain how to setup pulse signals ... and it solved the issue of my motors moving wrongly

I can't confirm if it solve the OLED shutdown issue ... as I short circuited the heater/thermistor ...and now am waiting for a new motherboard from biqu...

@kamilj the idea might be nice, but I really don't feel like biqu/btt should be in such a database
* there product are overall good and working as expected for a cheap price
* biqu is on an improvement with customer path (it is known their product are released as working but there will be newer upgraded version months after, so bugs are expected)
* they could do some more thorough job of debugging to only release final product but that would increase the price and the time to release (if that's what you're looking for, I think you might prefer some huge company that release product after years of developpment and not "on bleeding edge" ones)
* they are even providing some documentation (on which they might want to improve a bit, the english level got me at a loss some times )

and to end that, stating that "chinese products" need such a database, I disagree, you'd need a worldwide database as lot of scam are not from china. Referencing all, including real scam (like out of nowhere power generating system ie) would be a real huge amount of work.

Have a nice day

kamilj, 2020-09-02:
I think about Chinese products because the law does not protect the consumer in this particular case.

narosenberg, 2020-09-12:
Same problem here.  Most times the screen goes black as soon as I press the menu button, sometimes the characters are reversed left-right.  I have two of them (new) they both exhibit the same problem and are unusable.  I've contacted the vendor (BIQU) but based on others' messages, I'm not too hopeful.

Quas7, 2020-09-12:
does the issue also occure, if you disconnect the motor (power off before unplugging the motor)?
I suspect that some motors draw to much current and the resulting interference kills the OLED communication.

narosenberg, 2020-09-12:
The motor used is the one you provided as part of the kit.

 

I have just tried what you suggest – power off, disconnect the motor, power on.  Same result.

 

Neil

Quas7, 2020-09-12:
I have the same issue but much less severe than you guys. My extruder has this closed loop setup and after some printing the display gets corrupted or frozen.

ok, next try... try to disable "closed loop control" by switching dip switch 3 to off.
Dip 4 has to be on to run calibration.
there must be some source of interference.

What supply voltage do you feed to the boards? 12V or 24V?

fjrpilot05, 2020-09-13:
Similar issues as described here. (flakey OLED screen).  But not enough to prevent setup.  I am experiencing 2 deal breakers though.
1) random layer shifts
[image]

2) On my machine (Monoprice Maker Select Plus, w/SKR1.4 Turbo and S42B on x-axis only) when I do the math I should be at 160 steps/mm with 32 microsteps.  In order to get the x-axis dialed in I need it to be set for between 205 - 206 steps/mm
This is very odd to me.  The mechanics on the Monoprice dictate that it should be 160 as it is on the y-axis without the closed loop.  

Has anyone else who is using these experienced layer shift on long running jobs?  How about weird step count settings?

The layer shift problem is a deal breaker for me.  I thought this was impossible with closed loop during a normal print job (no interference, no hung up filament, no hitting the printer with a mallet.... :) )  But it appears that there is something flakey going on.

Regards

fjrpilot05, 2020-09-13:
@charchess 
With regard to the y-axis dancing around, Have you tried connecting up through the serial interface and play with the PID values?
This implementation of closed loop doesn't have any sort autotune. (The calibration is not for tuning PID values) so in order to get these tuned in for your implementation these values should be adjusted for each individual axis you use them on.
Regards,
fjrpilot05

Quas7, 2020-09-13:
> Similar issues as described here. (flakey OLED screen).  But not enough to prevent setup.  I am experiencing 2 deal breakers though.
> 1) random layer shifts
> [image]
>
> 2) On my machine (Monoprice Maker [...]

did you reduce the communication speed and increased pulse lenghts to the S42B?
in Marlin you can select the TB6600 external steppers to match the slower timing required by the S42B. It will show this behavior, if set to 4988 or TMC drivers. DRV8825 settings might work somehow.

Quas7, 2020-09-13:
I just noticed, that after flashing the compiled version of the firmware provided in this repository all the display issues vanished together with the bug "spin to zero position on re-enable".

Maybe you test and flash the boards with the latest version?
A few details how to do it can be found here: https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/3

If the issue persists for someone, there might be a hack by changing the GPIO speeds for the display communication.

oled IO in main.c (line 1215 onwards):
GPIO_InitStruct.Speed = LL_GPIO_SPEED_FREQ_HIGH; // switch to LL_GPIO_SPEED_FREQ_MEDIUM

It looks like the angular sensor (TLE5012B) is not interfaced via the same SPI bus but some other interface.
https://www.infineon.com/dgdl/Infineon-TLE5012B_Exxxx-DS-v02_01-EN.pdf?fileId=db3a304334fac4c601350f31c43c433f

Quas7, 2020-09-13:
For the shifted layers and hte change of communicaton parameters see also here: https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/14

You can test also with open-loop mode by switching pin 3 "off". If the layers still shift, it is not the PID.
For a wrongly calibrated PID I would normally expect ringing or undershoot but not a constant layer shift. That is why I think the S42B is thinking it is correct as it just did not catch all steps send by the mainboard.

Quas7, 2020-09-16:
after a few days of printin now, the display showed some erronous characters but it is still better than before the firmware update.

Quas7, 2020-09-22:
regarding the OLED there is a new finding for the initialization of the OLED here https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/16

Quas7, 2020-10-05:
PR is opened as a possible fix for the OLED issue. Testers welcome to approve.