bigtreetech/BIGTREETECH-S42B-V1.0 #9: Stepper drive in closed loop misbehaves...

Baltojikale, 2020-08-06:
Bought diy kits and installed on Geeetech a10 (printer similar to ender3 ). Calibrated and installed according TEaching tech video. Drivers in closed loop mode misbehave.. they count steps as red led lights up ( shows error) but driver doesnt move motors or moves periodically then once again dont move.... also tried changing stepping pulse width, and reflashed most current firmware from this repository. Uppon further inspection drivers works well in open loop but doesnt performs movement correctly in closed loop. 
What could be the problem??

mickeymicks, 2020-08-15:
I'm also seeing the same problem on a modified Duplicator 6 with MKS SGEN L board. 
Interestingly, this only happens to my Y motor. The one that I installed on my X motor works fine in both open and closed loop. 
I tried swapping the motors, connectors, etc and it's only happening on that one board.
For my case, the motor won't move at all and would just make some whiny noises.

Baltojikale, 2020-08-17:
at first was same only one axis misbehaved upon further experimentation both started to misbehave.

raimis170, 2020-08-24:
my does not work when i enable number 3 switch

Marcusbjol, 2020-09-04:
To get them to work, implement teaching techs corexy fix. Prepare to recalibrate the steps per mm tho.

Baltojikale, 2020-09-04:
Yes. i have implemented those fixes. but  it seems thats not the case.. few more test are left to perform but at 32 micro stepping setting for s42b i have topped out atmega2560 chip computational capabilities and thus layer shifting and closed loop misbehavior was due to slow motherboard . Just yesterday instaled SKR1.4 turbo (8times faster that gt2560 board). so today will be testing if problem still ramains.

So my suggestion to fix this issue with atmega2560 based boards is to reduce micros tepping to 16 or even 8 micro stepps per x and y axis.

Will update if problem remains with skr 1.4 turbo

Quas7, 2020-09-14:
could also be partly related to a too fast or boarderline communication timing. Would also explain the axis dependence @mickeymicks as the PCB routing differs for the steps signal.
See https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/14

Baltojikale, 2020-09-14:
So all tests passed and been printing non stop for a week with skr 1.4 Turbo with no layer shifts. Thought with linear advance feature enabled printing speeds kinda limited... And as i predicted 8 bit boards cant handle 32 microsteping on x /y  and spi on z/e0.  So in case of s42b misbehaving with 8 bit board first thing is to reduce microstepping.

closed 2020-09-14 by Baltojikale

mickeymicks, 2020-09-23:
> could also be partly related to a too fast or boarderline communication timing. Would also explain the axis dependence @mickeymicks as the PCB routing differs for the steps signal.
> See #14

I'll go try out the marlin settings in your link later. 
Will test it on a spare SKR 1.4 board I have, as for now, i've decided not to install s42b on my printer.
Just too much hassle and I feel it won't bring that much benefit in real world prints for my case at least.
Maybe some day, I will attempt to install it again.

Thanks!