bigtreetech/BIGTREETECH-S42B-V1.0 #13: Motor will not move!

tubaian, 2020-08-25:
I installed a S42B on the X-axis of one of my printers which has a SKR 1.4 turbo board and Marlin-2.0.5_7_7_20 installed and was working fine on a 2209 driver. Calibrated the S42B OK, commented out the X Driver info in Config.h, removed the references to stalled homing for the X-axis, changed the Minimum stepper pulse to 3 and Maximum stepper rate to 150000, all in Config.adv.h . Motor has faint vibration which changes slightly when I give a 'move X' command but the motor shaft does not turn. The only response to a move command is that the error value changes slightly and then goes back to its previous value (usually 0).
I have tried 2 different motors but neither work. Any ideas?

raimis170, 2020-08-31:
i have same. works only on calibration mode

AbeFM, 2020-10-02:
There's a note burried in the docs that says to set the minimum pulse width your printer is sending out - i.e. it might not be seeing the step commands. I believe this will require you to recompile Marlin, as I did.

I had other issues, I haven't tried going back, but it's something to look at.

[image]

tubaian, 2020-10-03:
Thanks AbeFM. In my initial post I did say that I had already tried altering the two items you specify but without any success. I haven't looked at changes in the s42b firmware sources though so I'll do that. I was going to try running them with an arduino to prove the motors are OK but I was so disappointed that they didn't just run 'out of the box' that I haven't got round to it yet..

AbeFM, 2020-10-05:
My bad - when I came back to this to answer it, I'd totally forgotten what I'd read.

You're still getting it to enable, right? Mine turned out to be missing the +5V line, then it saw the steps. If you're using a stock board I'm assuming that's there... I'm guessing the red "magnet too close" light isn't staying on (it's a second light)? I moved mine further, when trouble shooting, but it wasn't my issue and I feel is a little more unstable since.

With programming issues, it's dead in the water for now.

Quas7, 2020-10-05:
Could be realted to this observeration for certain bigger motors: https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/19
The biggest motor also did not turn outside calibration in closed loop (open-loop no issue) but for slow movements the display shows increasing angle deviations till "red light" at 36° delta.

tubaian, 2020-10-06:
Abe, I was trying to use the motors on a stock Bigtreetech 1.4 turbo board with Marlin 2. I have tried them with an arduino basic setup and they run fine provided I have a delay of at least 7 - 8ms between step pulses. I have spent hours poring over the firmware source code trying to find somewhere I could modify the 'sampling frequency of the pulse signal' but without success. Its galling that a manufacturer would supply a product that won't even work with their other related products.

Quas7, 2020-10-06:
@tubaian you mean 7-8ms delay between rising edge of the step-input (from the arduino) and the actual motion at the shaft?
I would not directly conclude that this delay is coming from the "sampling frequency of the pulses" but is more the nature of a stable PID control because the D value shoud exactly do that (differential or "dampening"). Maybe you set D to zero and measure (if oszillations do not make it impossible).

AbeFM, 2020-10-06:
I read it as the delay between the ending of one pulse and the start of the next pulse - too close together, you'll miss the pulse and count consecutive pulses as one long pulse. It's a form of aliasing.

Probably you could test it. I'd turn microstepping all the way down, but you don't need to. Have the printer move in a line at a shallow angle - say 1 degree - and/or turn steps per mm way down. If it does that move slowly (straightforward g-code, something like G1 X200 Y2 F10), the idea being you could spread the pulses out enough that they will get counted.

If things work as they do in my head, you'll see the spacing between pulses go up and the system function as expected, and that'll verify your suspicions.

mirecta, 2020-10-06:
I have same problem open loop works perfectly but close loop only one microstep and hang at Err: 35.99 and red stil shinig , calibration is ok . in Marlin 2 i choose TB6600 which has some frequencies and puse width. my board is BT GTR , i have logic analyzer can i help you ?

mirecta, 2020-10-06:
And i has MKS servo 42A 1.1 too and its work with same motors but i want BT S42S .

tubaian, 2020-10-06:
The arduino code I was using to test the motors is as follows..

int stepPin = 5;
int dirPin  = 6;
int  enPin  = 7;
 
int numstep = 4096;
unsigned long ldelay = 10;
unsigned long mDelay = 500;

void setup() {
  // put your setup code here, to run once:
    
     pinMode(stepPin,OUTPUT);

     pinMode(dirPin,OUTPUT);

     pinMode(enPin , OUTPUT);

     digitalWrite(dirPin, HIGH);
     digitalWrite( enPin , LOW);
}

void loop() {
  // put your main code here, to run repeatedly:

  for(int x = 0; x < numstep; x++) {

           digitalWrite(stepPin,HIGH);

           delayMicroseconds(ldelay);

           digitalWrite(stepPin,LOW);

           delayMicroseconds(ldelay);

        }

    delay(mDelay);
    digitalWrite(dirPin, LOW);

I found I could change 'ldelay' down to 7 on one motor and 5 on another before they refused to move.  Hope this makes sense to someone..

AbeFM, 2020-10-06:
> I have same problem open loop works perfectly but close loop only one microstep and hang at Err: 35.99 and red stil shinig , calibration is ok . in Marlin 2 i choose TB6600 which has some frequencies and puse width. my board is BT GTR , i have logic analyzer can i help you ?

Ah! I had this once. When I first put mine together, it worked fine, but then I spaced back the sensor board 1.2mm (printed washers), and a couple times it did this, especially if I bumped the wheel by hand. 

When I looked it up in the manual it says it might leave the light on like that if the magnet is too close. Anyway, I was about to say check magnet distance (supposed to be 2-4 mm), at least it's something to look at. Got what it's worth, my spacers are still in there and it only happened a couple times.

@tubaian - thanks for the code snippet, that looks great and it's cool you got a tester going like that. Those are with the same BigTreeTech board but different motors?

Quas7, 2020-10-06:
@tubaian Ok, so you experimentally found evidence that the timings from the S42B manual are still to fast for some batches?
Little sidequestion, did you use an Arduino Board with 5V or 3.3V controller?

And more imporant, did you already set the minimum stepper pulse width to 7 or more in configuration_adv for testing?
/**
 * Minimum stepper driver pulse width (in µs)
 *   0 : Smallest possible width the MCU can produce, compatible with TMC2xxx drivers
 *   0 : Minimum 500ns for LV8729, adjusted in stepper.h
 *   1 : Minimum for A4988 and A5984 stepper drivers
 *   2 : Minimum for DRV8825 stepper drivers
 *   3 : Minimum for TB6600 stepper drivers
 *  30 : Minimum for TB6560 stepper drivers
 *
 * Override the default value based on the driver type set in Configuration.h.
 */
**#define MINIMUM_STEPPER_PULSE 7**

mirecta, 2020-10-07:
i will try it today 
#define MINIMUM_STEPPER_PULSE 7

tubaian, 2020-10-07:
Quas7, I used an Arduino Uno with a single stepper breakout board. I had the MINIMUM_STEPPER_PULSE in Marlin set to 3 as Bigtech's page suggested - I'll try upping this to 7 later today. Thanks.

mirecta, 2020-10-07:
#define MINIMUM_STEPPER_PULSE 7 not work for me in close loop  stil freeze at 35.99 , in open loop still works perfectly

mirecta, 2020-10-07:
it is point to measure problem because in open loop it works even with pulse 2 and more,

tiagomaricate, 2020-11-03:
Is it solved? Same problem here.

Quas7, 2020-11-04:
> it is point to measure problem because in open loop it works even with pulse 2 and more,

comparing open-loop vs closed-loop might not work because the load on the controller is completely different in both modes. The controller does not have to manage the angle measurment and PID control and that might improve its response behavior for shorter pulse-length etc.

Quas7, 2020-11-04:
@tiagomaricate after testing with many different motors I can say, that the motor design itself is a strong factor. Try using a different nema17 model without any "magnetic parts" (test with a magnet) on the backside.

tiagomaricate, 2020-11-04:
> @tiagomaricate after testing with many different motors I can say, that the motor design itself is a strong factor. Try using a different nema17 model without any "magnetic parts" (test with a magnet) on the backside.

Indeed.
All my original nema have metallic plates/boxes on the backside.
So I tested with a new motor I bought a month ago. This motor has a plastic cover on the backside. And worked.
Now I'm checking if is best just to replace X and Y motor or replace all, to keep a standard configuration.

Quas7, 2020-11-04:
I am quite sure you can make it work with any motor, if you just move the board and encoder magnet very far away from the magnetic noise. I saw installations working fine on the "pulley side" instead of the backside.

NZ-Nitro, 2020-11-10:
Another person being driven mad by this problem, but also calibration accuracy as an example I give the axis a 10mm command and measure with verniers a 7.5 mm movement

- Tried changing the dip switches 1, 2

- Tried changing the steps per mm, this made no change on the error

- Tried changing the #define MINIMUM_STEPPER_PULSE no change on this error

- Reinstalled all the software on the computer no change

-

Quas7, 2020-11-10:
@NZ-Nitro maybe you checkout #24 and #19 
Very likely your nema17 motor is interferring with the magnetic encoder or your motor is wired differently than the board.

Mercury3D, 2021-06-04:
Hello,
Has anyone made any progress on this issue?  I am having the same hangup whereby the motor will do 1 microstep then hang at -36 degree.  Have tried all the switches, calibrations, marlin alterations and this problem is still occurring.  I am using S42B V2.0 full motor/sensor assemblies from BigTreeTech (meaning i have not altered spacing of the board or magnet.).  Trying to use these for the y axis of an Ender 5 Plus.  Even if i disconnect the motor from the frame and try moving the motor away from the machine it self it still will not move.  Open loop mode works fine, but as soon as i enter closed loop mode it seizes up.   Any help would be greatly appreciated have been trying to make this work for weeks!

ElekPat, 2021-06-04:
See this https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/33, it worked for me.