bigtreetech/BIGTREETECH-S42B-V1.0 #14: Steppers misbehaving workaround and steps per mm wrong

Marcusbjol, 2020-09-04:
When I recieived the 2 stepper motors, one did not move. I followed the fixes in teaching techs video by changing 2 lines in config_adv.h

#define MINIMUM_STEPPER_PULSE 3
#define MAXIMUM_STEPPER_RATE 150000

After implementing this, both steppers worked.

However, both have radically altered the steps per mm setting. Before the changes, they were set to 1280 with 16 microstepps. After, the movement was way off. By some experimentation, I was able to get it to print properly with y 204.95 and x 207.03 and 32 microsteps on both.

Reading config_adv.h, I will try 
#define MINIMUM_STEPPER_PULSE 1
#define MAXIMUM_STEPPER_RATE 500000
to match the driver.

BluPix, 2020-09-06:
Hi, i have discovered from manual that 32 microsteps means 8192 steps per 360° revolution.
So correct formula for steps per mm is: 8192 steps per revolution / (2mm GT2 belt * 20 tooth per revolution ).
example:
8192 / (2 * 16) = 256
8192 / (2 * 20) = 204.8

kamilj, 2020-09-09:
> When I recieived the 2 stepper motors, one did not move. I followed the fixes in teaching techs video by changing 2 lines in config_adv.h
> 
> #define MINIMUM_STEPPER_PULSE 3
> #define MAXIMUM_STEPPER_RATE 150000
> 

You do not read the documentation :P

42 Stepper Motor Closed Loop Driver Board User Guide
ShenZhen BigTree Technology CO.,LTD .

5、We found that when SKR V1.3 motherboard was used
to drive this driver, the motor did not rotate after the
motherboard gave S42B pulse.This is because the pulse
frequency was too fast and S42B could not be detected. At
present, there are two ways to modify:
1.Modify pulse width and pulse frequency in Marlin
firmware.
Details: Modify the high pulse width in configuration. adv,
# define MINIMUM_STEPPER_PULSE 3
# define MAXIMUM_STEPPER_RATE 150000
2.Modify the pulse signal sampling frequency in the S42B
firmware source code.

Marcusbjol, 2020-09-09:
The documentation is poorly written and translated (former tech documenter here). I did find a solution and documented it here for others to find.

Funny enough, hard coding 

#define MINIMUM_STEPPER_PULSE 1
#define MAXIMUM_STEPPER_RATE 500000

work as well.

fjrpilot05, 2020-09-13:
@Marcusbjol 
You are absolutely correct.  As it turns out it appears that your suggestion works better especially when the implementation is mixed with TMC2209's on the other axes.  I think this is so because these settings are global across all axes and the suggested settings in the manual (pulse 3 and rate 150000) severely hamper the 2209s.

Quas7, 2020-09-13:
in Marlin firmware you can just select the TB6600 drivers for the axis with the S42B in the Configuration.h Stepper Drivers section. Worked for me without issues with TMC2209 on X/Y/Z and only extruder with S42B.

This activates this timing profile just for the respective axis (see configration_adv.h in Marlin).

Minimum delay before and after setting the stepper DIR (in ns)
1500 : Minimum for TB6600 drivers (guess, no info in datasheet)

Minimum stepper driver pulse width (in µs)
3 : Minimum for TB6600 stepper drivers

Maximum stepping rate (in Hz) the stepper driver allows
150000  : Maximum for TB6600 stepper driver

fjrpilot05, 2020-09-14:
Thats an awesome suggestion.  I'm gonna try that right away.
Thank you

Marcusbjol, 2020-09-16:
Seeing odd behavior regardless of which stepper timing I set (tried their recommended settings). The first move of the X-axis moves very rapidly and then moves as expected. Their firmware needs some work.

Additionally, as stated before, their solution degrades performance with their TMC2130 stepper drivers as well.

They are responsible for providing compatible firmware for all their products (using BTT SKR 1.4 Turbo, BTT 2130 Drivers for Z+E0+E1, and BTT closed loop for X+Y). Incomplete product they need to update.

Quas7, 2020-09-16:
@Marcusbjol 
for the rapid movement at (re-)enabling you can check here (solved): https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/3

The timing can be set in Marlin per axis using the "TB6600" timing profile without impacting other stepper drivers.
Marlin documentation improvment for this is suggest already here: https://github.com/MarlinFirmware/Marlin/issues/19391

Marcusbjol, 2020-09-18:
Implementing that fix solved the issues. 

Thanks for the help guys.

closed 2020-09-18 by Marcusbjol

fjrpilot05, 2020-09-22:
> in Marlin firmware you can just select the TB6600 drivers for the axis with the S42B in the Configuration.h Stepper Drivers section. Worked for me without issues with TMC2209 on X/Y/Z and only extruder with S42B.
> 
> This activates this timing profile just for the respective axis (see configration_adv.h in Marlin).
> 
> Minimum delay before and after setting the stepper DIR (in ns)
> 1500 : Minimum for TB6600 drivers (guess, no info in datasheet)
> 
> Minimum stepper driver pulse width (in µs)
> 3 : Minimum for TB6600 stepper drivers
> 
> Maximum stepping rate (in Hz) the stepper driver allows
> 150000 : Maximum for TB6600 stepper driver
> 
> and I also activated this here as well for my TMCs
> /**
> 
> * Beta feature!
> * Create a 50/50 square wave step pulse optimal for stepper drivers.
>   */
>   #define SQUARE_WAVE_STEPPING

Hey Quas7,
I just wanted to circle back and Thank you for your suggestions.  I've got my setup pretty much to a happy place thanks to your suggestions.

Regards,
Kevin

Quas7, 2020-09-22:
Hey Kevin,

great that it worked out as intended and thanks for giving positive feedback. :)

Best regards,
Till

GIANNHSitia, 2021-02-14:
hello, i have same problem. i change X , Y axis driver with S42B 1.1 . Working but with wrong steps per mm..For examper: instead of moving 150mm this move 120mm.. with old driver work ok.. i add to configuration.h this lines.... but dont change nothing. how to fix this? sorry but my english is not good

#define X_DRIVER_TYPE TB6600
 #define Y_DRIVER_TYPE  TB6600