bigtreetech/BIGTREETECH-S42B-V1.0 #15: Dual steppers (z axis) wiring confusion

lousassole, 2020-09-09:
Good morning,

I cannot figure out how to wire dual Z to the S42B drivers on my SKR 1.4 turbo. It appears that i could use the adapter as well as the CLS pins on the board, but am not sure if this would cause issues. Would i be better off crimping a cable to use them in parallel, or daisy chain them? I just don't want to experiment with it and let the magic smoke out of the board or burn out a motor.

Marcusbjol, 2020-09-18:
I am no expert on this, so take what I say with a grain of salt.

It should make no difference (parallel or serial). There might not be enough signal strength for 2 closed-loop controllers to function. Burning out a motor is unlikely with the power not being enough, but the double draw of the motors might be (probably not tho).

Test it out and let us know how it goes.

Quas7, 2020-09-18:
maybe you add a drawing, what you want to wire in which way?

do you plan to go for serial Z before or after the S42B?
With "after S42B" I mean one motor with encoder and the second motor is just following via series connection of A1,A2,B1,B2 connection. This would be the only possibility to wire something in series, in my opinion. There is no power issue in this setup as the current stays same as for one motor.

In parallel config, all pins besides maybe GND and Motor-Vcc from the mainboard socket can be connected in parallel. 
GND and motor Vcc should be wired directly from a bigger connector than the socket as two motors in parallel might exeed the single pin current limit.
But, for a normal double Z axis the motors should not run above max 2x 800mA (RMS) which should be still fine for dupont style sockets and alike (max 2.5A up to 85C ambient, datasheets are hard to find on that rule of thumb though)

In my opinion, Z-hybrid steppers are quite an overkill as the repeatability and stability of Z with lead screws is normally a decade better than the X/Y belt axes. 
Maybe improves belt driven Z axis setups.

Marcusbjol, 2020-09-18:
Definatly mainboard -> 42B -> 42B.

I got my feet wet on smoothie and they have specific instructions on wiring external drivers in parallel (http://smoothieware.org/general-appendixes#multiple-drivers-in-parralel), so it should work in this setup.