bigtreetech/BIGTREETECH-S42B-V1.0 #18: The Y-axis is stuttering and acts sometimes like a guillotine

copperfield74, 2020-09-30:
Hi everybody, today I was able to implement the S42B on my Artillery Sidewinder X1 Y-axis.
I followed the steps that were shown on YT by Teaching Tech.
I put  switch 4 on and did the calibartion. After that I entered the microsteps (16) and also entered the mAh (1700).
But the first test wasn't successful because the hotbed only moved stutteringly and not fluently. Also I'm not able to move the bed all the way to the front. I thought that maybe the voltage isn't high enough so I raised it to 2100. But still the same problems. Also the Y motor is somehow "humming" as if it's somehow working once it moved. Sometimes when I've shutoff the motors, turn them back on and home Y it's flying like a guillotine. If I had my thumb between the plate and the camera mount it might have cut it off (or at least hurt me). See a video here:
https://youtu.be/W5ACd9wWcLg

I've calibrated it once again but still the same problem. I would say that the magnet it quite in the middle of the motor and the distance to the pcb is also okay because of the spacers that were delivered. I hope to get some help asap so that I can go on with printing.
PS: when Closed Loop is off (dipswitch 3) it moves fluently. My firmware is Marlin 1.1.9

AbeFM, 2020-10-02:
I believe this is an issue with not having the PID values tuned for your printer. What happens is the motor turns to position, the bed hasn't QUITE gotten there yet. The motor stops but the bed goes past, pulls the motor PAST the position, so the motor starts pulling the other way, and if there's a match between PID values and the response time of your bed, you'll get rocking. Mine was about ready to knock the printer over, shaking violently till I damped the oscillations with my hand.

I suspect the loop needs to be slowed down for massive loads, but I'm still trying to get reliable programming so I've yet to mess with it.

(watched video) I'm not sure why it's unevenly traveling, but I suspect that's what it is. Tightening belt will help a little, but it's not the RIGHT answer, and will wear belt.

copperfield74, 2020-10-02:
Hi AbeFM, thanks for your response.
I thought nobody was here anymore! ;-)
I've already changed the PID settings to see what happens. It got better, but I'm honest....I didn't know what I was doing! :-)
I couldn't find any good resource or tutorial describing PID! I don't even know what the values should do. It now moves a bit more fluent but it sounds as  if it was grinding. I changed the values to:
P:30
I:10
D: 300
But I don't know which value is the right one concerning my special problem. And it's a pain in the a** that I have to disassemble the S42B from the motor to connect it to my Laptop everytime to see if the changes are good or bad.
If PID is so important why is there no real documentation? Or more easy way to dial it in without the hassle of unmounting it every time. If I would get it to work I would buy a few more for my other printers but right now I'm sad that there is no help from BTT to solve these problems.

nhabes79, 2020-10-02:
I suggest changing the I value to 1. It wasn't even worth it to me to do any PID tuning until the I value was fixed. 10 is way to high. Also, make sure A4988 is the motor driver chosen in your firmware. I had some issues if any other motor driver was chosen in my firmware.

copperfield74, 2020-10-02:
> 
> 
> I suggest changing the I value to 1. It wasn't even worth it to me to do any PID tuning until the I value was fixed. 10 is way to high. Also, make sure A4988 is the motor driver chosen in your firmware. I had some issues if any other motor driver was chosen in my firmware.

Thanks for your reply! What do the different values stand for? What happens when I is changed. It would be great to understand what is happening. Has it to do with acceleration? Or what does the I stand for, or the P or the D! And yes I did change the driver to A4988 in the firmware. Please let me be your padawan...teach me please! :-)

Quas7, 2020-10-03:
I would recommend you select TB6600 as stepper drivers for S42B in Marlin. Even A4988 is still a too fast communication for the S42B. 
This change does not have a really big impact as long as you do not plan to go >500mm/s travel on 32usteps with 32bit controller boards.

copperfield74, 2020-10-04:
> 
> 
> I suggest changing the I value to 1. It wasn't even worth it to me to do any PID tuning until the I value was fixed. 10 is way to high. Also, make sure A4988 is the motor driver chosen in your firmware. I had some issues if any other motor driver was chosen in my firmware.

I've tried to go as low as I=1 but after that the Y axis didn't move anymore. I've changed it to 5 and now it's working again. It's not perfect but much better then before.

copperfield74, 2020-10-04:
> 
> 
> I would recommend you select TB6600 as stepper drivers for S42B in Marlin. Even A4988 is still a too fast communication for the S42B.
> This change does not have a really big impact as long as you do not plan to go >500mm/s travel on 32usteps with 32bit controller boards.

I still have my stock MKS Gen L V1.0 board in my printer so no 32bit board. Maybe I'll give it a try and change to TB6600.
Thanks for the hint.

megatech1966, 2020-11-10:
> TB6600.

Did you have any luck fixing your problem. I think I have the same problem. I’m waiting for my st-link to arrive so I can update the firmware and change the pid values.