bigtreetech/BIGTREETECH-S42B-V1.0 #19: Info: not all NEMA17 motors are compatible

Quas7, 2020-10-05:
After playing around with 5 different NEMA17 and 3 hybrid stepper DIY boards I figured that not all motors are "magnetically compatible" with the concept.

I did not bother reading out the field values with a hall probe or similar but just compared working setups with non-working ones by using "magnetic view foild".

The difference I observed is a change in sharpness of the N/S boarder (center) as well as the outer shape of the magnetic contour while rotating the shaft for the same airgap (no magnet tilt etc.). 
Seems, like the motor casing is magnetic or the permanent magnets from the inside of the motor interfere somehow. 

[image]
[image]
[image]

I have no direct proof but after seeing this I just added spacers between the motor shaft and the magent and additional spacers on the mounting screws. This got one of three previoulsy non-working setups working.

Therefore, if the S42B is not working for you maybe just try exchanging the NEMA17.
For me it was a 50:50 chance up to now with smaller motors (e.g 4047) working generally better than large ones.

Could, of course, have various different reasons but I just wanted to share in case someone made a similar observation.

raimis170, 2020-10-06:
what was your problem before ? did fix it?

Quas7, 2020-10-06:
for certain motors I did not get any movment. They were freewheeling till some magic lock-position and step commands showed angle deviation running up but no movment. Other motors showed movment but vibrated at certain angles quite heavily.

The spacers or a second axially magnetized magnet (reduces B field) did not fix all of them but for one it worked afterwards. It looks like it is not always a PID parameter issue but might also be an encoder or better a magnetic signal quality issue.

Others would have to test as well, if additional spacers below the encoder magnet get some combinations working.

Quas7, 2020-10-17:
not suprising but it looks like the spacer needs to be non-magnetic, e.g. 3D printed, and the magnet should at least be 50% outside of the motor housing.
With this improvment I got all motors to work that I tried so far (3 pcs).
[image]

MetalBeard72, 2020-10-24:
"For me it was a 50:50 chance up to now with **smaller motors (e.g 4047)** working generally better than large ones."

Forgive my ignorance. What does 4047 refer to? Physical size of the motor? Does higher amperage matter?

Quas7, 2020-10-24:
C4047 and similar are nema17 codings for common motors (https://reprap.org/wiki/NEMA_17_Stepper_motor)
For me this translates to 40 mm motor length.

Quas7, 2020-10-29:
I tested now a few more nema17 and found a totally incompatible one even with the magnet completely outside the shaft cavity. It does not move at all in close loop.

The main difference to the other motots seems to be a thinner aluminum backcover and a ball bearing close to the backside.
Magnets stick to the ball bearing though the backcover. This might be a first indication for compatibility.

[image]

MetalBeard72, 2020-10-29:
Hey, that looks promising.

MetalBeard72, 2020-10-30:
I switched to a motor that passed the magnet test and it worked! On another note, I think I fried one of my boards. It no longer powers up. (Unrelated to this post)

Brian-Zombait, 2020-11-08:
> The main difference to the other motots seems to be a thinner aluminum backcover and a ball bearing close to the backside.
> Magnets stick to the ball bearing though the backcover. This might be a first indication for compatibility.
>

Okay, so more spacers for the board and insert a non magnetic spacer between the shaft and magnet. That means longer screws, too.

Good thing I used only a whisp of gel super glue. Should be able to pry it off.

IhorNehrutsa, 2020-11-11:
Thank you, Till Quas7.
[image]
Calibrated well after the proper direction of the motor rotation (Issue #33).
I see your Pull requests.
May you change the firmware to see degrees during the calibration process?

Quas7, 2020-11-11:
I am quite sure, that the "calibration" process does not actually result in anything close to absoulte degrees. It can only be a relative degrees given by the motor precision of something like 1.8° +/-5%.
This also means, that before and during calibration the sensor has no clue what the actual angle is, as it is only referencing every new full step (+1.8°) to the readout of magnetic field of the sensor.

But, what should be doable is to output the number of steps already done and total steps needed like a progress indication.
Also, one could do a verification run after calibration as the MKS servo42b does.

Q5Grafx, 2020-11-29:
I tried your solution by doing this. i needed a 5mm x 2mm spacer for the magnet. i found plastic spacers at granger but only had minimal contact on the magnet and shaft because of the 3.2mm hole in the center. then i typed into a search 5mm discs. which brought me to etsy i ordered 5mm x 3mm stainless beads which have a 1mm hole and a different option of 5mm copper stamp cuts. i went with the copper. i had to use 4 layers. for centering i have 8mm od with a 5mm id spacer and use more magnets as a handle. what i noticed is the copper offset the magnetic field and other magnets would not line up with it on center anymore. But its due to magnetic flux from the copper and actually helped. for standoffs i went to grainger and found 6mm m3 spacers and they fit inside the countersinks on the motor and give the perfect distance of about 3mm from magnet to sensor. after that the setup was effortless and had no problems with it. Now all i need is an easy way to set the PID for tuning because i hate seriel port communication. but the system is functioning however i cant see where they get their step counts. a 8 microstep it says 2048 pulses but on my cnc its only 512 but then its as accurate as i have the capability to measure. i decided to use the 16 step and bumper the setting to 1024 and my pins are 1 on 2 off 3 on 4 off and its accurate 10cm = 10 cm.