bigtreetech/BIGTREETECH-S42B-V1.0 #23: TMC CONNECTION ERROR

davidcgu, 2020-10-10:
Good night guys,

I just tried to install this on E axis while originally I had working an TMC2209 on UART with no issue.

I'm able to power up and complete calibration succesfully however I'm getting TMC connection error.

I'm I send M122 from pronter I see that issue is coming from this slot, E0.  If I move back and install again the TMC communicatoin error is gone.

I have no clue on how to solve it do you have some idea?  

From other side, I see a 4 dip switch onboard and have no clue what is the purpose of this, I can find out anything about it on manual, do you have some document where is explained the purpose of those switches?

Thanks in advance

davidcgu, 2020-10-10:
trying to find the way to delete the post, any info regarding the funcion of the 4 position dip switch?

Thanks in advance

Quas7, 2020-10-11:
did you already do te steps in the manual?

dip 4 is write protection of calibration data.
dip 3 is closed loop vs open loop mode.
dip 1,2 set microstep resolution.

set 4 to off and select your preferred microsteps with 1,2.
Them run calibration without any load on the motor (detach belt).

Also A4988 is not the right setting in Marlin. Chosing TB6600 is more  appropriate for the signal timings.

davidcgu, 2020-10-11:
> did you already do te steps in the manual?
>
> dip 4 is write protection of calibration data.
> dip 3 is closed loop vs open loop mode.
> dip 1,2 set microstep resolution.
>
> set 4 to off and select your [...]

Thanks for your reply,

I read all PDF and I think I have not seen any clear steps to follow... what steps are you speaking about?

My actual marlin config changes are:

#define E0_DRIVER_TYPE A5984
#define INVERT_E0_DIR true
#define MAXIMUM_STEPPER_RATE 150000
#define MINIMUM_STEPPER_PULSE 3

Calibration process is completed succesfully as mentioned previosuly, on open loops i turns fine is when moving to close loop when I facing the problem, I could try to change to TB6600 but not sure if this will change anything,,,,

From other side, should I insert jumpers on the board to select the microstepping mode?

Regards

swanepoeljan, 2020-10-11:
Sounds like it could be an issue with the magnet installation, and that the distance between the magnet and the sensor needs some adjustment. Maybe experiment with adding/removing some of the small spacers. Think the manual said the sweet spot is somewhere between 2mm and 4mm.

Quas7, 2020-10-17:
@davidcgu you could take a look at #19 as this solved it for me for all stepper motors I tried.

Brian-Zombait, 2020-11-08:
> @davidcgu you could take a look at #19 as this solved it for me for all stepper motors I tried.
> 

I am experiencing the same problem as the OP. My machine is an Anycubic Mega S.

I've already updated the settings in the motherboard firmware to reflect the pulse min 3, and the max rate to 150,00. The default driver on the board is A4988. Also, motherboard firmware has been updated to the latest version of Merlin using the Anycubic Mega file set from their website to ensure its set up properly for my machine. Note - it works fine without the S42B controllers. 

On addition, I have tested the values, via the oled, of the micro steps from 16 firm to 4 with no change in closed-loop mode. Again, open-loop mode works fine.

I have not attempted to change PID values, and have no experience doing so. All firmware changes and updates have been done via Visual Studio. 

The distance between the magnet and sensor is 2.62mm, based on my measurements with my trusty calipers.

Any help with this would be appreciated, as my whole reason for getting the S42B controllers was for the closed-loop function. 

[My motherboard is an ANYCUBIC TriGorilla Integrated Main Board Compatible Mega2560 & RAMPS1.4 Combo Control Board 4 Layers PCB Controller Board for RepRap Mendel Prusa I3.](https://www.amazon.com/dp/B079GRJH6V/ref=cm_sw_r_cp_apa_fabc_SqaQFb8Z1X3H7?_encoding=UTF8&psc=1)