bigtreetech/BIGTREETECH-S42B-V1.0 #25: X axis dancing

Sekai1942, 2020-10-23:
Hi. i just installed s42b on my diy printer and did the calibration and it works but the x axis when i try to move it it just goes back and forward really fast.I think the whole asamble is to large in weight and it overshouts when moving then try to compensate then it overshouts in reverse and that becames a loop.
I try to lower the accelaration but the same thing.
The Y axis does the same but not every time (as shown in video).
And when i turn off the motors in the interface below, the y axis i can move it but the x axis the motor stays energized but there is no compensation meaning the board is not active for close-loop.
Any ideea how to solve this?
https://drive.google.com/file/d/1P0TRuBtJh8re5k-FxD4gFDRKnzX-ddsW/view?usp=sharing

Quas7, 2020-10-23:
Did you calibrate without load (belt detached)? That would be my first shot as imprecise calibration kills the PID loop quite easy.

If the calibration is fine without load I would check for the PID settings of the S42B as they likely need tuning for heavy loads. Can be done either via serial control (see the manual) or via flashing the firmware directly (my prefered way).
Oszillation should be either too much I or P or to little D. Instead of I=10 a few use I=2 or even I=1. Just reduce I until you get no oszillation. If this is not possible, reduce P as well.

I am not sure, how far Jan is already for his fork to get PID tuning more user friendly: https://github.com/swanepoeljan/TrueStep

Maybe you check the other issues in this repo as well as there are some tunings required for Marlin 2.0.x as well.

Sekai1942, 2020-10-25:
You are right. i calibrated without load.I calibrated the y axis with load and it looks ok now but the x axis i have no way to reach out to the buttons. i will have to serial control it.
thanks.

Sekai1942, 2020-10-25:
btw how to upload the firmware?

Quas7, 2020-10-25:
for the flashing of the FW you will need an stlink dongle (clone). More details are here:

https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/3#issuecomment-691197566

Sekai1942, 2020-10-28:
> for the flashing of the FW you will need an stlink dongle (clone). More details are here:
> 
> [#3 (comment)](https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/3#issuecomment-691197566)

Thx for the replay. I order the dongle about a month ago for different project.
The current problem i have is i cannot modify the pod from serial.
It gives me a response back but when i try to change the parameter, it gives back crc sum error.
What i am doing wrong?

Quas7, 2020-10-28:
@Sekai1942 I never used the serial interface and always flashed my changes directly. So, I can not help much on this one but there are other issues discussing on this here, I think.

swanepoeljan, 2020-10-28:
@Sekai1942 Did you see [this](https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/blob/master/Serial%20data%20transfer%20protocol.pdf) document. It explains how to calculate the checksum.

Sekai1942, 2020-11-09:
yes i found out how to modify the parameters.

swanepoeljan, 2020-11-09:
In the meantime I have added support for the original firmware in my [terminal program](https://github.com/swanepoeljan/TrueStep/blob/master/TrueStepTerminal.zip) which you could also use for changing the PID gains.

robtheminnie, 2021-01-04:
Does this firmware also work with the s42b V2.0 or is it only suitable for the V1.0?

swanepoeljan, 2021-01-04:
Since V2.0 uses the STM32F103 and probably also use a different pin out it will most likely not work. As for the terminal program, I am not sure, if they kept their protocol the same it could probably work.

robtheminnie, 2021-01-04:
@swanepoeljan, thanks for the info.  I commented in another issue, #32 , about having the same problem.  The user fixed the issue by updating to your firmware fork.  However, as I have V2.0 boards, this isn't possible.  Are you familiar with the #32, and behaving indicated, with vertical lines and ringing.  I'm assuming it's incorrect PID gains.  Any thoughts?

swanepoeljan, 2021-01-05:
Haven't seen that post, glad to hear it worked for him. I would also try playing with the gains to see if it helps, maybe he can share his printer model and gains that worked for him.