bigtreetech/BIGTREETECH-S42B-V1.0 #26: motors loud/screeching

lousassole, 2020-10-23:
I am at a loss here. I have installed these drivers/kits on my x and y axis, and on both axes when holding position they emit a loud whine. When moving it becomes more of a screech. I have tried loosening my belts, which did not help much. I have ensured the voltages are correct as well, and that everything is mounted flush.
The printer still runs well, the screeching is nearly unbearable though.
Is there anything i'm missing that fixes this?

Quas7, 2020-10-23:
do you have a datasheet for your nema17 motors? E.g. the inductance, resitance and current rating?

lousassole, 2020-10-24:
Brand: Creality 3D
Item name: RepRap 42 Stepper Motor 
Item number: 42-34
Step angle: 1.8 degrees
Nominal Voltage: 4.83V
Current Rating: 1.5 (A)
Rated speed: 1-1000 (rpm)
Rated torque: 0.4 (NM)
Ambient Temperatuar: -20℃~+50℃
Length: 34mm

The printer runs off 24v, mainboard is a BTT SKR 1.4 turbo on marlin 2.0.x bugix
they are the default x/y motors for the ender 3. As far as i'm aware, they should be compatible.

Quas7, 2020-10-24:
ok, and changing the motor current does change the screaming frequency?
I had loud motors only, if I wired the phases incorrectly (A1,A2,B1,B2 S42B output but some motors are A1 B1 A2 B2 input). But with wrong wiring they should move very erratic and you get successfull prints, right?

lousassole, 2020-10-25:
prints are successful (even had a successful 16hr job last night), and changing voltages seems to have no effect as far as i can tell. It seems to be that whenever there is any strain placed on the axle of the motor it produces this screeching sound.

Quas7, 2020-10-25:
I think, your motors are a bit on the high side with their inductance (>7mH maybe). 
The required voltage to drive 1.5A is quite high which means there is more wire with its series resistance on the motor coils and this results in a higher inductance. 
High inductance makes motors reluctant to change direction (or accelerate) and that should hit the decay mode of the stepper driver stage and might result in audible noise. Normally, higher or lower supply voltage should change the pitch.

In other words, you could try nema17 with <4mH or just around 3V for 1.5A and around 40Ncm torque.

https://reprap.org/wiki/NEMA_17_Stepper_motor

lousassole, 2020-10-27:
i'll play with the voltages some more. I'm trying to keep the motors as cool as possible though, so i may end up having to switch to a different motor.