bigtreetech/BIGTREETECH-S42B-V1.0 #28: Board no longer powers up

MetalBeard72, 2020-10-30:
I'm afraid I fried my board. I have been very careful to power down before making any connection or switch changes. Now the green LED no longer turns on.  Just on the off chance that I can resurrect this thing, Has anyone else saved one of these?

Quas7, 2020-10-31:
can you still communicate with the controller via the SWD interface (e.g. still program it?), if powered only via the SWD interface?
If this does not work anymore, I would trash it.

Quas7, 2020-10-31:
Maybe also check with magnification that there is no bent pin/short or any other metal flicker sticking somewhere.
Reassembling the boards often might have thown some metal splinters from the screwing process on the board...

dzid26, 2021-03-06:
I wonder if BEMF could have kill it. 

Because when I drive the motor from the output fast enough, the MCU resets. @Quas7  Do you have idea why is it restarting? 

The input voltage is stable and current is never reversed.  - Hmm, maybe that's the key, because I think the board have reverse polarity protection diode. So if that diode doesn't allow the current to flow in reverse, the internal voltage will rise high. Though, I would expect the linear regulator to heatup, and no MCU reset. 

....Hah, while writing this I run it with a drill for longer, while it was reseting. It seems the A4950 didn't handle BEMF. Mosfet are shorted and it is drawing 0.4A continuously now.  Bummer. 
But MCU still works. 

Anyway, I still don't udnerstand why it was resetting.

I guess for now I am going to stick with MKS42B which doesn't have reverse polarity diode. I also put clamping diode on supply lines to protect for overvoltage.

dzid26, 2021-04-06:
Ok, so I think it resets (and then burns) from BEMF, due to bad location of the clamp diode.  
[image] 
@bigtreetech  Maybe V2.1 could be improved.