bigtreetech/BIGTREETECH-S42B-V1.0 #29: Missing License file and source files.

JustGitting, 2020-11-01:
Hi BTT,

Seems BigTreeTech's major positive selling point is that sells "Open Source Hardware". Giving users freedom to modify, repair and create new applications with BTT hardware. However, this goal is not backed up by the appropriate license or source files (gerber PCB design files, etc) in this repo, see #5 for example.

Please add an appropriate Open-Hardware License (e.g. OSHW) and provide the hardware source files (e.g. gerber, netlists, etc files) to this great project.

Best wishes.

GithubSimba, 2020-12-08:
@JustGitting Hello. Do you have any reply from BIGTREETECH?

JustGitting, 2020-12-08:
Hi @GithubSimba,

Nothing from @bigtreetech, which is rather disappointing.

Unfortunately there are companies that claim to produce "Open Source Hardware" because either they don't really understanding the meaning of OSH or they are being deceptive. Simply adding a few PDFs and some software does not make it "Open Source Hardware". OSH should allow users to reproduce their own device without the need for proprietary tools or software (in general), hence the meaning of Freedom.

@bigtreetech please update your repo with the appropriate OSH license and hardware source files as soon as possible or stop pretending to be for OSH.