bigtreetech/BIGTREETECH-S42B-V1.0 #30: How to update the firmware?

tiagomaricate, 2020-11-03:
Hi people!
I was reading tome topics related but the documentation itself is very minimum.
I'd like to know if there is some kind of tutorial or similar to help me on how to update the S42b.
So far, the only thing I know is the need for an adapter.

Quas7, 2020-11-03:
A starting point can be found here: https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/3#issuecomment-691197566
A working adapter would be a stlink v2 clone, or any other stm32 programmer that can be found on Amazon.

Setting up the development environment consiting of Visual Studio Code and platform.io can be found on youtube, e.g. https://www.youtube.com/watch?v=cmHQxd_qGl8

Hope this gets your running.

tiagomaricate, 2020-11-03:
> A starting point can be found here: [#3 (comment)](https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/3#issuecomment-691197566)
> A working adapter would be a stlink v2 clone, or any other stm32 programmer that can be found on Amazon.
> 
> Setting up the development environment consiting of Visual Studio Code and platform.io can be found on youtube, e.g. https://www.youtube.com/watch?v=cmHQxd_qGl8
> 
> Hope this gets your running.

Tks!!!
I will give a check.

imperator-maximus, 2021-03-04:
Platform.io was not able to upload firmware here (" OpenOCD init failed"). Solution: use STM32 ST-LINK Utility. Open firmware there and upload it via "Target"-"Program"

vc, 2021-03-22:
After uploading [TrueStep](https://github.com/vc/TrueStep) firmware via ST-LinkUtility, you can upload firmware via serial port using [TrueStepTerminal NG](https://github.com/vc/TrueStep/tree/master/utils/TrueStepTerminal)

paulrbeers81, 2021-05-03:
@vc is TrueStep compatible with S42B v1.1?  I assume so since the processor is the same and I THINK it's just a tweak to the layout, but maybe not?

vc, 2021-05-03:
> @vc is TrueStep compatible with S42B v1.1? I assume so since the processor is the same and I THINK it's just a tweak to the layout, but maybe not?

You are right. TrueStep firmware specifically for S42B v1.1

ravx4, 2021-05-25:
Hello everyone,
After reading topics about True Stepper, I decided to give it a try. 
I have managed to upload firmware.bin via ST-Link utility and it's working fine.
However, when I try to connect via true step terminal, I get "could not connect" message no matter what baudrate is set. 
Also I have tried connecting my S42b v1.0 to TS-terminal on 2 different laptops and pc (win 10 and win7) and still got the same error.
Can anyone give me any clue what could be the reason of this error?

vc, 2021-05-25:
> Can anyone give me any clue what could be the reason of this error?

I think you'd better ask a question in the repository from which you took the firmware and the TS-terminal program. There are already several varieties of these components.

ravx4, 2021-05-25:
Thank you for reply @vc, 
although I don't quite understand how to ask in original repository (I'm using files from @swanepoeljan repo) as there's no possibility to open an issue there.

vc, 2021-06-01:
> although I don't quite understand how to ask in original repository (I'm using files from @swanepoeljan repo) as there's no possibility to open an issue there.

Yes, you are right, the issue section has been closed in @swanepoeljan's repository.
You can try using my build of s42b firmware and TrueStepNG software. I forked from @swanepoeljan repository and did some tweaks. If it is interesting, you can look here: https://github.com/vc/TrueStep

ravx4, 2021-06-13:
Ok, I'll try your repo, maybe can manage to get board comunicating with my pc,
Big thanks again :)