bigtreetech/BIGTREETECH-S42B-V1.0 #32: Poor quality prints

megatech1966, 2020-11-06:
Hello all. I installed 2 BTT S42B V1 on to my x and y. Calibrated ok. Uncommented x and y in configuration.h. Y was playing up. Wasn’t moving all the way to 0 on my bed. Changed 2 lines in configuration_adv.h 
#define MAXIMUM_STEPPER_PULSE 3

#define MAXIMUM_STEPPER_RATE 150000

Now y is working ok
Had to change x steps to 100 and y steps to 100

The problem is I get poor quality cube. Banding and horrible ringing. The last nice one was with my TMC2130s. 
My max acceleration is 500
Any help would be appreciated 😃
[image]
[image]
[image]

megatech1966, 2020-11-06:
[image]

S42B using open loop (switch 3 off)

robtheminnie, 2021-01-04:
I have a similar issue with mine.  I have these fitted to x and y on an ender3, and now have the same ringing/vertical lines shown in the initial post.
Have you found a fix?

megatech1966, 2021-01-04:
I updated the firmware from a fork of the btt firmware

robtheminnie, 2021-01-04:
> I updated the firmware from a fork of the btt firmware

Do you have a link?

Was it from here?

https://github.com/bigtreetech/BIGTREETECH-Stepper-Motor-Driver/tree/0814d4a0371aa818ba08abeb214c3f460904cde9/S42B/firmware

megatech1966, 2021-01-04:
https://github.com/swanepoeljan/TrueStep

robtheminnie, 2021-01-04:
Ah, thanks.  Also, just seen you are using the V1.0 board.  I have V2.0 boards so that fork won't work unfortunately.
Thanks for the info though :) I might get in touch with the writer of that fork and see if he has any input.

GIANNHSitia, 2021-02-15:
I have the same problem, have you found a solution? I have S42b 1.1 to X and Y axis

[image]

megatech1966, 2021-02-15:
I have a solution for version 1.0

GIANNHSitia, 2021-02-15:
how to fix it?  not working to 1.1v?

megatech1966, 2024-04-07:
This issue should be closed.

closed 2024-04-07 by megatech1966