bigtreetech/BIGTREETECH-S42B-V1.0 #33: Direction of motor rotation during calibration?

IhorNehrutsa, 2020-11-11:
My motor was worked well only in an open-loop mode (DIP 3 is OFF). 
After Calibration (when DIP 3 is ON) when the motor shaft was rotated manually, the Err: parameter changed near +/- 1 degree. 
I pushed the magnet out of the motor housing as described in issue #19. That not helped.

Closed-loop mode began work, only when I swapped two wires of one motor coil. That changed the direction of motor rotation during the calibration process. 
So I think there is a bug in the calibration process which requires a certain direction of motor rotation. First CCW, then CW (View from the shaft, not from the driver board).

Quas7, 2020-11-11:
Interessting. I will test this next weekend.

ElekPat, 2020-11-12:
Thank you very much, I could not get my motor to run in closed loop (i tried all previous suggestions) until I tried this: IT WORKS NOW !!!

dzid26, 2021-01-19:
Or one can change DirPin direction in the menu