bigtreetech/BIGTREETECH-S42B-V1.0 #34: Endstop issues

MasterBPM, 2020-11-30:
Hi,

i have to say i am very unpleased with this Product so far. Bought 5 pre-assembled s24b v1.2 from biqu and I am now troubleshooting since 14h. So on all 5 Steppers the magnet has different dimension so I had  to disassemble all of them and measure, combining spacers to fit correct distance between magnet and sensor. Also 1 Display provided a short (lucky I had spares). 1 Motor is still not running in closed loop, but in open. So much "fun" for 32€ each. Anyway ...

I am running an skr 1.4 Turbo with 5 of these and newest marlin Bugfix. The only thing I really think I can't figure out myself is the problem I had on all endstops in closed loop.

---------

X and Y - hits endstop, stepper stops but  raise up from 0.04° to ~35° error/misalignment and stepper shows red led till I press reset button or reset printer. No issues in open loop.

BL touch clone for z - axis turns up a bit (5 mm) should then pull down pin and motor steps down to heatbed, but instead it doesn't pull down the pin but also here error count goes up to ~35° , red led turns up and only reset pin or printer reboot fix it.

With other motors and tmc 2209 everything is running fine, also in open loop everything is as it should be. But I need the closed loop feature otherwise this product is absolutely useless for me.

I hope someone could help.

Cheers,

MBPM

Quas7, 2020-11-30:
is the s42b stepper driver timing already set to slow?

MasterBPM, 2020-11-30:
Hi,

Iam not 100% sure what you mean. If you mean pulse settings I tried it commented, 1/500000 and 3/150000 in marlin conf adv.

Cheers,

rafaljot, 2020-12-08:
I am very disappointed with the lack of a feature from the nano_stepper project. Especially the virtual endstop.
https://github.com/Misfittech/nano_stepper

"
**errorlimit**
Gets set the maximum number of degrees of error that is acceptable, any posistioning error about the error limit will assert the **error pin**, when error pin is set as error output. 
For example:
:>errorlimit 1.8 
Will set the error limit to 1.8 degrees. 
"
I think it could be the best option to manage endstops.