bigtreetech/BIGTREETECH-S42B-V1.0 #35: S42B 2.0 Closed Loop Not Working.

rennwaffen, 2020-12-01:
Just installed the new S42B 2.0, and closed loop doesn't seem to want to work. I'm waiting on a response from BTT (which takes a little while) so I figured I'd ask here in the meantime. I also have a V1.1 on my X axis that works flawlessly. Any idea as to what the issue could be? Notice the deg. value not changing as the axis moves.
Info:
Calibration went fine (Supposedly)
DIP 3 is set to ON
DIP 1, 2, 4 are set to OFF
32 Microstepping
Tried currents as high as 1000mA (2000 setting) with no luck.
Board: SKR 1.4 Turbo

[image]

dzid26, 2021-01-19:
Swap one one of the phase polarity:
https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/33#issue-740859328

N6MQD, 2021-01-28:
There is another issue posted that may be your issue.  DIP 2 is for Closed Loop mode.  DIP 1 is for Calibration.  DIP 3 and 4 are for subdivision.