bigtreetech/BIGTREETECH-S42B-V1.0 #36: S42B_v2.0 won't enter calibration mode

pizzabola, 2020-12-16:
Hey I'm writing this be cause the lack of clear instructions on the BTT github site.
My set up:
BTT SKRv1.3
Stepper 17HS4023
S42B_V2.0 board
This is what I've done so far:
I installed the hardware S42B and the oled display on my x axis stepper motor. After that, I remove my TMC2208 driver from the skr 1.3 and installed the cable from the S42B and put it in place (in UART mode). On the S42B I flipped all the switches to off but switch #4 which remained on according to the pdf from BTT to enter calibration mode. 
On marlin, in the configuration.h,  I comment out //#define X_DRIVER_TYPE  TMC2208 in order to make the stepper driver to default "A4988". In configuration_adv.h I enabled "#define MINIMUM_STEPPER_PULSE 3" and also enabled "#define MAXIMUM_STEPPER_RATE 150000" changing the values from 250000 to 150000. I compiled everything on visual studio code and uploaded the sketch to the skr1.3. I turned on the printer and everything seemed working just right but when I try to calibrate the stepper using the oled display of the S42B nothing happens. I mean, the s42b turns on but no blinking lights or motor movement at all. From the lcd display of the printer I was able to make the steppers move no problem. The only problem is that the S42B won't enter calibration mode. I'm I missing some step to set up closed loop control right? Any help would be pretty much appreciated thanks

Q5Grafx, 2020-12-17:
I have the same with a couple of mine. at first they would go to calibrate whenever i wanted them to. one day they stopped. My machine is working flawlessly but i cant do a calibrate. Im running 3 of these on 60mm steppers and my machine was kinda close before but i could never get exact dimensions. when i added these its dead on and repeatable. With mine i had issues with a few motors and ill explain what i did so maybe it will help you. I set mine to activate high so i had to set $4=1. still was getting issues with start and stops but the software thinking the motor was moving but it wasnt. pin 1 and 2 are for microstep settings, make sure they are set to match the setting on the oled screen. i set the power to the power level the motor was rated at and not doubling it like others have said because the power is halved. I also had to add a non magnetic spacer behund the magnet to raise it slightly outside of the aluminum motor housing. for these i found 5mm copper stamped discs on etsy. it took 4 to get about 2mm height increase. then put on a 5mm spacer to lift the board to get about 3mm space between the magnet and the sensor. when i turned on my machine after this last mod i couldnt get into calibrate on 2 of the three motors, but when testing actual calibration against a caliper i am with .001" which for me is good enough. in addition the stepper motors are completely silent now where they had never been anything close to silent. I ran a few test jobs last weekend and all went well, all dimensions were dead on and cuts were clean so now its time to make christmas gifts for the grandkids. I forgot to mention if i set pin 4 to on and pin 3 to off i can get into calibrate on those 2 motors. but not when closed loop mode on and calibrate pin turned off. the third motor doesnt really care what settings are and it will calibrate any time i want it to. the third motor is new though so i wonder if it will stop going to calibrate after awhile.  Hope something in this rant helps.

GhostlyCrowd, 2021-01-19:
Did you guys ever solve this i have 2 brand new ones and neither will enter calibrate mode, the rest of the menu works.

jcmental, 2022-11-02:
Hi Sorry to bring up an old thread, but I've just installed 2 of these and neither will accept any function i select from the menu, I have 2 S42B V2.0 I can scroll through the menu but when I press the enter button nothing happens. I have all dip switches to off except #4