bigtreetech/BIGTREETECH-S42B-V1.0 #37: schematics??? and 0.9deg motor support?

jlmxyz, 2020-12-20:
Hi,
on the documentation, it's said we have to cross some motor control lines and not some others... on my motor there is only wire that goes outside... I assumed the wire color were the same as the provide cable... but there is nothing that happens when entering calibration except a bad noise... my motor is a 0.9degree motor, obviously the calibration will fail because a whole rotation won't be performed...

1. I need the schematics of the board to now how the drivers are connected to the motor control interface
2. I don't get what the purpose of the switches on the center of the board, again a schematic will greatly help
3. I will have to reflash the board in order to support my 0.9 motor... I read that there is only a way through the JTAG inteface on P4 connector?

thanks and regards

Quas7, 2021-01-13:
I think, you should maybe read a summary on NEMA17 motor wiring and working principle. You have to connect A and A' (coil A) as well as B and B' (Coil B) of the motor correctly to the 4-pin connector of the board. 
The wire color is no indication here as it is not standardized at all. 

Here are the requested informations, if that can help you.

1. it is hidden in a not very convient way here: https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/blob/master/Item-Pinmap.PDF
[image]

2. read the manual on page 7 and page 9 for the dip switch meaning (schematics will not help you because they are just inputs to the firmware): https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/blob/master/42%20Stepper%20Motor%20Closed%20Loop%20Driver%20Control%20Board%20Manual.pdf

3. it is the so called SWD interface. Pinout see pinmap above. Just make sure, the motor is not enabled during programming as this messes with the signals. You will need a stlink v2 or similar to program. Good luck.

pctj101, 2021-01-18:
Hello!

I looked at https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/blob/master/Item-Pinmap.PDF

I see your info, but it does not contain the part number, or part values (such as 330 Ohm).  Is the part list available?

Thank you!

Quas7, 2021-01-19:
there is no BOM (part list) provided, afaik.

jlmxyz, 2021-02-03:
> I think, you should maybe read a summary on NEMA17 motor wiring and working principle. You have to connect A and A' (coil A) as well as B and B' (Coil B) of the motor correctly to the 4-pin connector of the board.

well, in fact this don't help... on https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/blob/master/Item-Pinmap.PDF there is no information about where A and A' and B and B' are located.... there is also nothing printed on the pcb.... a schematic would have gave me the answer... but no... so I will have to try and guess...