bigtreetech/BIGTREETECH-S42B-V1.0 #41: How to Update firmware

adamfilip, 2021-01-17:
How Do you connect the S42B v1.1 board to USB?
it has no USB ports

IF i need to use another device.. what pins does it use?

Quas7, 2021-01-17:
dpublicate of #30

JWest1969, 2021-02-26:
I too have questions about this. There was no programming cable supplied with the unit. There is no description other than an image of 4 wires plugged into the header pins on the board. I have no serial port on my computer, just usb, so I have no clue what to do. My installation went smoothly until I ran a test print and the result was very ugly print quality. I hope someone will chime in and show/tell about the actual device needed to connect to the board to access parameters that are not available through the provided user interface of the oled screen and buttons. It sucks that the device doesn't work correctly out of the box and that myself as an end user of the product has to become a firmware programming expert in order to make it work on my machine! If I turn dip switch 3 off (open loop mode) prints look great! With dip switch 3 turned on (closed loop mode) prints look terrible.

NZ-Nitro, 2021-03-21:
The information supplied by BTT is pretty shit, did you know between 1.1 and 2.0 the button functions reverse?