bigtreetech/BIGTREETECH-S42B-V1.0 #45: Manual and CAN bus control for V2.0 board?

mfxpyro1, 2021-01-25:
Where can I find the manual and details on how to use CAN bus to control the S42B V2.0 board ? I can't seem to find it anywhere.

Wayne2580, 2021-01-29:
I am interested in this as well

Frontpage32, 2021-02-19:
I too am interested. I have gotten some info but it's depressing if correct. Seems it only works on a specific duet board and doesn't work on any BTT boards at this time. (Please correct me if I am wrong) So it's is basically useless till BTT makes a board or you switch to duet well due to the cost that is not going to happen anytime soon. 
This is the board I was directed to but seems duet has 2 or 3 that can use CAN BUS. 

https://duet3d.dozuki.com/Wiki/Duet_3_Mainboard_6HC_Wiring_Diagram

Sunbuzz, 2021-02-19:
I am also interested.

NZ-Nitro, 2021-03-21:
i CANT FIND ANY INFORMATION anywhere ON THE s42b v2.0

dzid26, 2022-08-12:
Use [Intellistep ](https://github.com/CAP1Sup/Intellistep) firmware.