bigtreetech/BIGTREETECH-S42B-V1.0 #46: S42b 1.1 after calibration write 357.7 degrees

GIANNHSitia, 2021-01-25:
Hello boys, i bought two S42B 1.1 for my printer. I test both with a nema17 and as after calibration, on screen write rpm 0, error 0 and degress 357.7. I tried many times to do it again calibration but the result is the same. I watched some presentation videos in youtube and saw after calibration on screen write 0 degress. my question is, is it defective or let me fix it. this video shows the problem please check - and the second problem is, when moving the stepper motor, the speed on the screen is 0. some write something

https://youtu.be/cgP7WpYkTbE

GhostlyCrowd, 2021-01-26:
In the video you never shut the calibration Dip switch off after you calibrate and reboot and turn the Closed loop dip switch on. So first i would  be trying that.

GIANNHSitia, 2021-01-26:
Hello, i try, with close loop dip switch OFF, after calibration and i press reset button, on screen the degrees is 0. After this, i turn off the power suppy and i turn the dio switch calibration OFF and the Close loop ON, turn again the power supply on and again write 357.6 degrees. I try and without 3d printer mainboad and is same. I have 2 S42b and present the same.
-note, that it works normally I just can not understand what 357.6 degrees means, and if it is normal

I attachment a video
https://youtu.be/GVUeH1f0JPE

GhostlyCrowd, 2021-01-26:
> Hello, i try, with close loop dip switch OFF, after calibration and i press reset button, on screen the degrees is 0. After this, i turn off the power suppy and i turn the dio switch calibration OFF [...]

Try Turning Dip 2 on for Closed loop and 1, 3 and 4 off does that work? 

Some of these boards seem to have the dips reversed, my V2 are both like this.

GIANNHSitia, 2021-01-26:
The dip switch 1 and 2 is the microsteps, 3 is close loop on/off and dip switch 4 is calibration. If dip switch 4 is off i cant calibration. 1 on and 2 off is 16.microstep ( i try 1 and 2 on but is same). Do you can a calibration video with v2 board???

GhostlyCrowd, 2021-01-26:
> The dip switch 1 and 2 is the microsteps, 3 is close loop on/off and dip switch 4 is calibration. If dip switch 4 is off i cant calibration. 1 on and 2 off is 16.microstep ( i try 1 and 2 on but is same). Do you can a calibration video with v2 board???

IF dip 4 off you cannot calibrate then your dips are not transverse.  what happens if you send it a movement command does the red light turn on and the motor fail to move?

GIANNHSitia, 2021-01-26:
No no work normal, the red light is off. If i try move the motor with my hand, this return to last position.the close loop working. Mabe the 357.6 degress is the calibration accuracy. 360 degress -357.6 = 2.4/200 step=0.012 error accuracy