bigtreetech/BIGTREETECH-S42B-V1.0 #47: s42b v1.1 frame err! ch340g connection

linewallker, 2021-01-27:
Hello to all,
  I have some issue when I connect s42b v1.1 board via ch340g usb to ttl driver, no baud rate works and still getting only Frame Err! when I am on serial monitor.

Even connecting with TrueStep Terminal is not working and I am still getting this error again and again, but TS Terminal is reading the settings I need but when I change I can not write back with error message that something went wrong.

Does anyboody have any clue what to do to get it work ? I need to change PID settings due oscilation os X axis arm
[image]
[image]

when I change to 115200 baud I am getting only symbols
[image]

please help guys

GhostlyCrowd, 2021-01-27:
TX on the USB-TTL goes to RX on the S42B and RX on the USBTTL goes to TX on the S42B. GND goes to GND and don't connect  3.3v or 5v on the USB-TTL, Power the S42B via the Printers power. I have a cheap ch340g usb to ttl and this works for me.

linewallker, 2021-01-28:
Now I can connect and read data with no problem, but I can not write Back anything with error something went wrong
[image]

linewallker, 2021-02-22:
Thanks for your a swer, I will try tomorow and let you know if It worked for me. If I should be honest, i have conected only by Ch340 to usb without printer so this is possible the answer for me 😉Odesláno z mého telefonu Huawei-------- Původní zpráva --------Od: GhostlyCrowd <notifications@github.com>Datum: st 27. led 2021 21:54Komu: "bigtreetech/BIGTREETECH-S42B-V1.0" <BIGTREETECH-S42B-V1.0@noreply.github.com>Kopie: linewallker <linewallker@centrum.cz>, Author <author@noreply.github.com>Předmět: Re: [bigtreetech/BIGTREETECH-S42B-V1.0] s42b v1.1 frame err! ch340g connection (#47)
TX on the USB-TTL goes to RX on the S42B and RX on the USBTTL goes to TX on the S42B. GND goes to GND and don't connect  3.3v or 5v on the USB-TTL, Power the S42B via the Printers power. I have a cheap ch340g usb to ttl and this works for me.

linewallker, 2021-03-01:
Ok, so after some time I must say that serial terminál is working solution but the P I D tuning is a large pain in my ass right now.
I do not know how to count the correct value so I am trying manuál way.

After about 1000 tests i am exactly where I started, axis still oscilate and create wawes in the plot even whe i am not able to setect vibration. 

Do somebody know how to find correct values for P. I. D.?

linewallker, 2021-03-04:
It was not grbl so we can close it..

closed 2021-03-04 by linewallker