bigtreetech/BIGTREETECH-S42B-V1.0 #51: S42B-V2.0 firmware rebuild

xhejtman, 2021-03-19:
Hi I tried to rebuild firmware for S42B-V2.0 version. It works, however, built firmware has only 35kB while pre-build firmware 2.0 has 54kB. This is ok, but the OLED display refreshes very slowly when flashed rebuilt firmware. If I flash pre-built, refresh speed is normal. Moreover, platformio settings for V2.0 has actually settings for V1.2.

Any ideas how to rebuild fast V2.0 firmware?

GhostlyCrowd, 2021-03-19:
The firmware the posted for V2.0 is incomplete and not what they shipped on the devices and does not work correctly. 

They have ignored all requests ive sent to update the repository.

xhejtman, 2021-03-19:
Thanks, is there any firmware that can be used for V2.0? I need to rebuild to add 0.9 stepper support.

GhostlyCrowd, 2021-03-19:
a fellow by the name @CapSup1 is doing a rewrite in his free time its not working yet

CAP1Sup, 2021-03-20:
As of right now, the LCD is working (minor bug left with buttons, I'm going to fix it today). The serial and CAN bus should also be working, but I haven't personally tested them yet. As for the motor, I had issues with the encoder, so that's also on the list of things to fix. I'm getting close, I'm hoping to have it finished by the end of either this month or the next. I've just been busy lately working on my Robotics team.

xhejtman, 2021-03-21:
Thanks, great work. Should 0.9 motor work as well, if sources are tweaked a bit (mostly sed 200/400).

CAP1Sup, 2021-03-21:
@xhejtman I've already been working on implementing that, the motor calculations should accept basically any step angle you like once I iron out the kinks. I'll create a release once I finish with the stepper testing. I'm shooting for the end of the next month or two