bigtreetech/BIGTREETECH-S42B-V1.0 #54: what are the dip switches for????

jlmxyz, 2021-03-31:
I'm a bit puzzled... 
on the doc https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/blob/master/42%20Stepper%20Motor%20Closed%20Loop%20Driver%20Control%20Board%20Manual.pdf, there is written that 2 firsts switches are for microstepping :
1 1 = 4
0 1 = 8
1 0 = 16
0 0 = 32

but on the screen we can select the microstepping....
so.... what is what for????? does the switches overdrive the selection in the menu? does the menu overdrive the selection of the swithes?  is the menu the settings for the controlBoard->ClosedLoopControler settings? or are the swtiches for controlBoard->ClosedLoopControler settings?  or is the menu for ClosedLoopControler->Motor? or are the switches for ClosedLoopControler->Motor????

I'm lost, the ClosedLoopControler calibrate, but nothing move... while it move with makerbase's servo57B (by the way I'm using S57B CLC, not sure this is the right place to ask, but there is no bigtreetech-S57B-V2.0 repository....)

thanks and regards

GhostlyCrowd, 2021-03-31:
the dip logic is inverted and BTT is too incompetent to update their source or manuals.  Dip 1 On is to Calibrate Dip 2 on is Closed loop Dips 3 and 4 are for micro stepping.

NZ-Nitro, 2021-03-31:
I think they provide dip switches for if you don't use the screen, also look at my post about accuracy

jlmxyz, 2021-04-01:
> 
> 
> the dip logic is inverted and BTT is too incompetent to update their source or manuals. Dip 1 On is to Calibrate Dip 2 on is Closed loop Dips 3 and 4 are for micro stepping.

what does it means "DIP 1 ON is to CALIBRATE" 

1. DIP1 has to be ON for calibration, OFF for normal use?
2. DIP1 can be left ON for calibration AND normal use?
3. DIP1 had to be OFF to prevent recalibration, has no impact for normal use?

DIPS 3 and 4 apply on what micro stepping? motor microstepping? control micro stepping????
eg are MotorStep = LCDConfiguredMicroStep * CommandStepping / DipswitchMicroStepping?
or MotorStep = CommandStep * DipSwitchMicroStepping????

really thoses DIP switch just brings pain in the ASS.... why adding something that cost more and can be configured with the LCD???