bigtreetech/BIGTREETECH-S42B-V1.0 #59: BIGTREETECH S42C

NZ-Nitro, 2021-11-21:
New model?

Anyone tried it?

andrew-cowell, 2021-12-05:
i have it.
installed on a voron 2.4 ON X Y.
when moving in just x axis or y axis they drift the toolhead out of place.
its like the motors loses torque when not used then eventually corrects it self as it has reached it destination.
also when you do Z lift and or Bed mesh.
the z is already moving down before they are on the destination they should be.
the manuals are crap there is no in depth  information.
the lcd menu lacks  options in my opinion.
screws should be 100mm long so they fit EVERY type of stepper motor 
no install information but its kinda straight forward.
The only information to make these puppy's work is on marlin.
no clear documentation how to write to the uart port.
the alu heat and mounting block is then again A SUPER AWESOME feature.
just needs a top plate now .
2000rpm they say 
can not get them to run fast in anyway at this moment.
Running them on klipper.
En these needs a auto brake on them to work there.

TheRevenantStar, 2021-12-30:
Can we get some docs on this model?

andrew-cowell, 2021-12-31:
on my voron its very cleary this servo pcb is under preforming.
even when current is high !
and you can not twist the pulley.
once you install them and they get a command.
they do drift the firmware is crap at this moment.
but am sure this can be fixed also they say they are uart capable!
PLEASE !!! better documentation For klipper and or any other firmware.
as is!!!
these are 100% unuseable unless!!!
YOU WANT TO PRINT SLOWER THEN you did before.

TheRevenantStar, 2021-12-31:
I haven't noticed any drift, or slow movements.
Mine is setup on the Ender 5+'s X-axis (I have yet to replace a defective one to setup the Y-axis) as of 2 hours ago.
The device performs exactly as advertised, once I figured out how to get the thing to calibrate.

In the end, speed and accuracy are typically inversely related. You can't expect high-speed printing and perfect accuracy, even from a CLS; though a CLS can help you approach perfection.

_On a personal note, and I hate to pull this card, maybe use more complete and understandable sentences. If anyone at BTT is using Google Translate, fragments and bad grammar make it harder to understand you._

andrew-cowell, 2021-12-31:
On a personal note, and I hate to pull this card, maybe use more complete and understandable sentences. If anyone at BTT is using Google Translate, fragments and bad grammar make it harder to understand you.

Zal ik het dan gewoon volledig in het Nederlands doen ??
Zodat niemand van jullie het verstaat :)
Sorry dit mijn leesblindheid u probleem is.
en het is niet omdat dat jij nu maar op 1 werkt!
dat het niet waar is wat ik zeg.
ik heb er wel degelijk 2 werkende.
En ze driften dus wel.
heb dit al bewezen met geposte video.
Dus als ik van u was zou ik u getrokken kaart maar terug steken.
Want u ganse woord salade.
is tegen strijdig met u claims.
en iemand bij btt reageren ???
ik ben al meer dan een maand nu bezig om een reactie te krijgen.
NIKS!!!
Dus wat als ze dit lezen met een vertaler.
And the grammar is not the problem :)

The-Distal, 2022-01-02:
https://github.com/bigtreetech/BIGTREETECH-Stepper-Motor-Driver/

You can find the S42C documentation here. I still can't get mine to work (fails calibration) and their support hasn't gotten back to me about issues... but at least you can see their code. The doc isn't great, but it's better than nothing.

mink007, 2022-06-15:
> I haven't noticed any drift, or slow movements. Mine is setup on the Ender 5+'s X-axis (I have yet to replace a defective one to setup the Y-axis) as of 2 hours ago. The device performs exactly as advertised, once I figured out how to get the thing to calibrate.

Any tips and tricks on how to calibrate ? Other than just plugging in the wires and following the instructions on the OLED?

reynhardtstef, 2022-08-01:
Does the firmware for the S42c even exist publically? I have also reached out to their support team but no one has contacted me yet.